neilh10/ModularSensors #7: Convert from ini reader to json reader with Arduino jsom

SRGDamia1, 2019-09-10:
We're doing a workshop with the Open Lab at Oregon State.  http://www.open-sensing.org/#intro. They've developed (with a much, much bigger team) a system they call LOOM to do many of the things ModularSensors does and quite a lot more. https://github.com/OPEnSLab-OSU/Loom/

Anyway, they're using a configuration file setup similar to yours but based on json. I think the json is easier to read and the ArduinoJSON library looks to be excellent. JSON is also a more standard structure than ini files. Would you be interested in looking into using JSON?

SRGDamia1, 2019-09-10:
I feel kind of bad asking because I still haven't really dug into your ini implementation, but I am quite impressed by them functionality that the Open Lab has developed.  (They also have snazzy things like a GUI to write code and control boards remotely.)

neilh10, 2020-04-09:
gosh @RGDamia1 - I don't know what happened here but I hadn't noticed this. 
 I did discover LOOM after reading about their AGU presentation last year.
 Configuration info could  work as JSON. The JSON parsers that i've worked on tend to take up more ram and build a tree in ram 1st. I find JSON is for machine2machine and that it is more intolerant of human errors.  
The .ini parsing that I'm working with is fairly simple and doesn't use much ram.   
I suppose I'm really focused on the objective of being able to scale Mayfly's, how to have the geographical information coded into the SD. 
What I think could really be useful is what would it take to automate the MMW Sensors output into a format that can be copied over onto the Mayfly SD file that, could then be automatically read in. As ever though ram useage would need to be checked first before JSON should be considered.

neilh10, 2020-04-09:
(i'm hoping I get notified of changes here - but not seeing notifications for my own entries)
The current method is to minimize code changes for main classes.  
I'm creating a branch of my fork that will have a straightforward example I'm talking over with Matt Barney
The .ini file concept is simple with [sections] and name=value pairs. So not as flexible as JSON, but easier for human reading.

Not as visible, but I define a local file to control the build with defaults ms_cfg.h

It uses the entry file .cpp/.ino  with setup() to add
#define SerialStd Serial
#include "iniHandler.h"

with the following in setup()
#ifdef USE_MS_SD_INI
    //Set up SD card access
    Serial.println(F("---parseIni "));
    dataLogger.parseIniSd(configIniID_def,inihUnhandledFn);
    Serial.println(F("\n\n---parseIni complete "));
#endif //USE_MS_SD_INI

Then LoggerBase.cpp is extended for processing the SD file in 
https://github.com/neilh10/ModularSensors/blob/release1/src/LoggerBaseExtCpp.h

and then found keys, in the iniHandler is to match to a key. 
https://github.com/neilh10/ModularSensors/blob/release1/src/iniHandler.h
this comes from    https://github.com/benhoyt/inih

Key matching from a specific _UUID,  to variables, requires code, and is going to be the same I think for .ini or a .JSON.
As far as I can see a basic JSON would be something like this,
https://arduinojson.org/v6/example/config/
then it would still need to mapp UUIDs from MMW to sensor UUIDs.

neilh10, 2020-06-04:
Closed no further discussion

closed 2020-06-04 by neilh10