neilh10/ModularSensors #12: SDI-12 Init

neilh10, 2020-07-14:
The SDI-12 is a software driver using a port bit as a line driver.
For the Mayfly mega1284, this makes the port bit line driver drive the line between 0 and 3.3V. The SDI-12 specification is to drive the line between 0 & 5V.  For a number of SDI-12 instruments receiving 0-3.3V works for them and they transmit back at ~ 5V.  The mega1284 ports are only specified to 3.3V+0.7 ~ but it still works with no immediate harm to the processor.
The blue line below shows the actual voltage readings on the SDI-12 Data line, with Yellow pointer showing Tx from mega1284, and Red pointer showing Rx from SDI-12 device.
[image]
And the the power on sequence
[image]
or showing power on 
[image]

To meet the SDI-12 specification requires a level shifter, and a surge suppressor is also recommended. A circuit with the SwV set to +5V, and the 3V3 created from the +5V. 
[image]
 
So the SDI-12 data levels are now activated (SPACE) when switched Vcc is turned on. Then it appears that there is a glitch (MARK) when the ports are init. (just my guess so far)
[image]

 I've not managed to find the source of it, so documenting it here

HAO156, 2022-04-11:
I think it's because of adding a pull-up resistor on the external bus while the IO is set as an input.

neilh10, 2022-04-11:
Hello @HAO156 yes I agree, however some where in the software is the initialization that creates the pull down glitch. 
I've purchased a https://vegetronix.com/Products/SDI-TRANS-SENSOR24/ to do more testing and also integrate CRC however I've also moved to RS485 as being more reliable on the target instrument I'm looking at.