neilh10/ModularSensors #98: Cell Network Option

neilh10, 2022-01-15:
The Digi LTE-CATM1 XB3-C-A2 supports an option for type of Network, and others may as well.

NB-IoT only (primary) is likely to be lowest battery usage, but will need testing.
This may need to be configured per network type,  or perhaps even for a comparison of power usage.

LTE-M with NB-IoT fallback(DEFAULT)
NB-IoT with LTE-M fallback  
LTE-M only
NB-IoT only
Digi XBee3 Cellular LTE-M_NB-IoT  90002258revV 2021Jan.pdf  Pg  189
N# (Preferred Network Technology)  ~ Set at Initialization in:
DigiXBeeCellularTransparent::extraModemSetup
gsmModem.sendAT(GF("N#"),0);    LTE-M/NB-IoT

SRGDamia1, 2022-03-31:
I removed that from the modem setup and moved it to example code back in #306.

neilh10, 2022-03-31:
@SRGDamia1 thanks for the comment. 
I saw this in  ModularSensors\examples\DRWI_DigiLTE  from line #378 - and so thought I should at some point check it out.

I've been meaning also to write up that specific examples\DRWI_DigiLTE  didn't work for me.

I was working with a SSU Capstone Engineering Project group, and suggesting using it as a way of checking out whether a REVX Verizon SIM on Digi LTE-CATM1 XB3-C-A2 was working - and we couldn't get the Verizon service to work, it failed.  REVX is https://dataplans.digikey.com  and I have three others working in the field.

I had to go back to manufacturing default of the LTE, bring up XCTU and erase it on a Digi development board, and then plug it in to a working system of mine to verify that the Verizon SIM service was working.

I've been meaning to track it.    Should I write up examples\DRWI_DigiLTE  failure as an issue?

SRGDamia1, 2022-03-31:
Oh, OK, I understand.

It's not worth making another issue for that example.  That example is a left-over and I plan to remove it as soon as @s-hicks2 gives me permission.  We've basically given up on the Digi LTE Bee's for DRWI folks because we've had ridiculously high failure rates and they haven't been consistently available.  Nearly 100% of DRWI users have Hologram SIM cards and needed AT&T friendly settings.  The example was specifically for their use.

In the gigantic menu example there is a section in the setup function which shows how to set the carrier profile and network technology and lists out the possible options for both.  If anywhere, that would be the best place to make changes or notes.

I also remember there being comments either here or (more likely) on the forum about settings to use to get the Digi LTE-M Bee's to connect better with Verizon.